ozel/DIY_particle_detector #3: 9V battery does not fit into recommended housing

Matthias32, 2020-06-20:
Please see the image. Or am I doing something wrong?
[image]

ozel, 2020-06-20:
It looks correct in principle, the fit is a bit tight and different 9V battery brands come at varying thickness (I've seen up to 17 mm). Have you tried already to close the lied? There is a bit of space/cut-out within the lid part as well.
Also, C5 should be rather flat like in this picture (labeled C6 in that picture - previous PCB version):
[image]
In case you've soldered C5 already upright, you can still change that by carefully unsoldering the capacitor with tweezers and afterwards bending its legs a bit.

I am using mostly 9.6V NIMH accumulators which are only 15.5 mm thick and come in a black plastic case. That helps with avoiding short circuits - insulation tape around the metal encased battery should help otherwise. In general, the pins of the electronic components must be cut as short as possible to make room for the battery.
Also the case must not necessarily close 100%, it is fine if there is a thin slit, just don't forget to apply the flexible sealing rubber that comes with it in order to have everything light-tight.

It helps if the PCB is thinner than the usual standard 1.6 mm thickness, rather 1 or 0.8 mm. That is of course too late in your case. I'll add another Wiki page about using those tiny die-cast enclosures.
Really consider going with the thinner rechargeable 9V/9.6V NIMH accumulators. It also helps avoiding all the battery waste and they're quite cheap these days!

Matthias32, 2020-06-21:
Thank you a lot for your quick response!

Yes, I tried to close the lid and was not able to do so. I have measured it and you’re right my battery has 17 mm thickness. The depth of the housing till my 1.6 mm PCB is about 16 mm (without seal). So this will barely ever fit.

I will try these 9.6V NIMH, lay down the C5 and cut the component wires a bit more. Maybe I also try to use an 0603 100 nF for C5.

ozel, 2020-06-21:
Yes, using an SMD capacitor for C5 is another good idea, if you have one. And really cut the OpAmp pins short, every fraction of a millimetre counts. ;-) Really do remember to insulate the battery's metal case from all the sharp cut pins... a piece of hard plastic foil cold work, too.

The lid itself has a pocket of a bit more than 2 mm of depth. If you allow for an additional slit between bottom case and lid, you should gain almost those 2 mm of headroom above the bottom case.
Such a slit may look not that nice but is completely fine for operation. The Faraday cage is closed enough as long as the four screws are in place because they establish the main low-conductivity path. Since aluminium always grows a thin non-conductive oxide layer on all surfaces, it doesn't really mater if the bottom case and the lid touch each other on the edges or not for what concerns shielding against potentially disturbing electromagnetic radiation. But you have to make sure it's closed enough to block all light, therefore the white rubber seal should not be omitted and needs to be put in place into the small trench of the lid.

Also, I am routing the connection wires always around the sides touching the curved edges of the battery, such that they're not in the way and adding to the overall thickness.
The on/off switch by the way just fits on top of the diodes, mounted in another hole on the short side of the bottom case. It is then somewhat in the way of the diodes, but that doesn't affect  much the sensitivity in case of the electron-detector variant.
I definitely need to take more pictures of this kind of build, but don't have a readily built on lying around - they're currently all on loan and used by others.

Matthias32, 2020-06-21:
Thank you a lot for the additional recomendations. I will try it out.

By the way: Can you please pass me a link of the 9V battery you are using so I can make sure it is the 15.5 mm one?

ozel, 2020-06-21:
[GP17R8H = 8.4 V version](https://www.gpbmindustry.com/product-page/gp17r8h)
[GP17R9H = 9.6 V version](https://ind.gpbatteries.com/pub/media/uploads/pdf/rechargeable_NiMH/by_series/standard_series_cylindrical/GP17R9H.pdf)

I prefer the latter for its higher voltage and did the alpha energy calibrations with that one. Since you're building the electron variant anyway, it doesn't mater at all.
Both seem to come in the same black plastic case.

ozel, 2020-06-21:
Thanks, @Matthias32 for reminding me about missing info related to the small aluminum diecast enclosures.
I've added two new wiki pages [Enclosures](https://github.com/ozel/DIY_particle_detector/wiki/Enclosures) and [Batteries](https://github.com/ozel/DIY_particle_detector/wiki/Batteries) so I'll close this now. Feel free to open the issue again or create a new one if you have more questions.
 
And please post a picture of your finished detector online so I can update the [Gallery](https://github.com/ozel/DIY_particle_detector/wiki/Gallery)! :D

closed 2020-06-21 by ozel