ozel/DIY_particle_detector #4: Wiki pictures

Matthias32, 2020-07-07:
[image]
[image]
[image]
[image]
[image]

ozel, 2020-07-11:
Thanks a bunch, @Matthias32! I've added some to the wiki gallery and will embed the rest as examples for a nice radiation window solution in the other wiki pages.  

I am leaving this issue open so that others will hopefully be encouraged to upload their own build pictures as well! :-)

ozel, 2020-07-11:
adding Malte's build picture from https://www.mikrocontroller.net/topic/495968#6327455 to make it public
[image]

Matthias32, 2020-09-22:
[image]
Mobile version

Matthias32, 2020-10-10:
Disassembly of foil capacitors
[image]
[image]
[image]

closed 2020-12-09 by ozel