ozel/DIY_particle_detector #7: Mounting the PCB

SteveF161, 2020-08-15:
Apologies for a noob question. My son and I have been building the electron detector as a lockdown project. We are ready to mount into a nice metal tin but we are not sure of best way to do so. I'm thinking some sort of motherboard/PCB stand-off but I'm not sure what size these need to be. Are the PCB holes for M2 , M2.5 or M3 stand-offs? Is metal better than plastic? Seems you only buy standoffs in quantities of 100+ so I'm trying to avoid expensive trial and error!

(Incidentally, the board proved really easy to build for a couple of inexperienced people. I haven't soldered anything other than audio cables for many years. My son has done only 3 or 4 joints in school. We both suffer from hypermobility of the finger, which doesn't help.  So, we checked every joint and all looks good so far. We plan to build the alpha detector next if this works OK.)

Matthias32, 2020-08-15:
I would recommend M3 metal. It should be possible to order less than 100 pcs. Just check on Amazon there are also some collections like this one:

[https://www.amazon.de/AMZANDY-NEW-Abstandshalter-Abstandsbolzen-Motherboard-120/dp/B0825XY6VD/ref=sr_1_1_sspa?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=m3+abstandsbolzen&qid=1597494331&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEyVEw3UkU2SFRYSDlFJmVuY3J5cHRlZElkPUEwMTU3ODI0MlNEOTVBWE9QUko2SiZlbmNyeXB0ZWRBZElkPUEwOTIyODg2MUZJMERMN0dNUjRaNiZ3aWRnZXROYW1lPXNwX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=](url)

Depends on from where you want to order. I was checking the german Amazon page for example.

SteveF161, 2020-08-15:
Many thanks for a swift response. That particular one isn't available on UK Amazon but I found some similar ones. and not too expensive.

ozel, 2020-08-19:
Hey @SteveF161, thanks for your feedback! Motherboard stand-off screws (or 'distance bolts' ?) could work but at least older ones often use non-metric threading. I use M3 parts made from metal since the holes and board edges are connected to the circuit ground. Using metal parts can improve the noise to signal performance - depending on the case. And using one of the holes should be enough.

I heard RS online ships for free in the UK also to private persons, their "M3 PCB spacer" should be a good choice:
https://uk.rs-online.com/web/p/standoffs/1613631/
(plus one M3 screw and a nut per detector)

Please do not hesitate to post pictures of both board sides if something doesn't work, it is rather easy to spot soldering problems visually in this design. And definitely consider posting some pictures of the finished detector [here](https://github.com/ozel/DIY_particle_detector/issues/4) so I can add them to the wiki Gallery. :-)

I'll keep this issue open in case others have more mounting related questions. So far this is covered here in the wiki:
https://github.com/ozel/DIY_particle_detector/wiki/Assembly-Instructions

closed 2020-12-09 by ozel