ozel/DIY_particle_detector #11: Alpha particle pulses

Matthias32, 2020-09-22:
Dear Oliver,
my beta detector works well but I'm not sure about my alpha version. Besides some small pulses (beta i guess) there are also some lager ones which goes also a lot in the positive voltage area. I also see often these "double pulses". Is this normal for the alpha sensor variant?
[image]
[image]

_Originally posted by @Matthias32 in https://github.com/ozel/DIY_particle_detector/issues/10#issuecomment-696834811_

ozel, 2020-09-23:
The second image looks quite good. 1-2 ms pulse width as expected on the downward-facing part.
The full amplitude is not shown, but if it is larger than -8 mV it should be a nice alpha particle pulse. Judged only by the large overshoot on the top, it probably is one. This amplitude threshold level of -8 mV between electrons and alphas (see main Readme file) is valid for short BNC cables,  1 megaOhm input resistance and a 1:1 attenuation setting at the oscilloscope channel.

The regular artifact in the first image looks like EM noise or stray light - it seems to be close to 50 Hz. Could you post a picture of your whole measurement setup and also show the alpha radiation window if you made one?

Matthias32, 2020-09-23:
This was the setup. The alpa radiation windows is just a hole in the bottom of the case.
[image]

ozel, 2020-09-25:
From this picture, it is difficult to give advice. I would need to see how the case itself looks like, including the radiation window etc. 
Did you cover the diode at least with foil like described here: https://github.com/ozel/DIY_particle_detector/wiki/Diodes#pin-diodes?
Even with that foil, it may still need to be used in only dark environments or with a piece of black cloth covering the detector plus specimen. Did you try black cloth instead of the aluminum foil?

Honestly, the aluminum foil will do not much. It is usually not even good in blocking light due to tiny cracks or small openings/folds close to the table surface. And for shielding low-frequency EM radiation - ***which is absolutely required here*** - it also does not work since aluminum always builds up a non-conductive oxide layer. It simply cannot be used to form a proper Faraday cage.
It may shield some high-frequency (GHz) EM radiation but that is usually not a problem with the DIY particle detectors and their low-pass characteristic.

Matthias32, 2020-09-30:
Thank you a lot. This is the detector below the aluminium foil. 
[image]
I will try to block the light with a black towell instead of aluminium foil.

ozel, 2020-10-05:
Hi Matthias,
that looks great in principle. But you either have to cover the diode itself or the hole with a thin metal foil (e.g. from an old foil capacitor) as shown in the first picture here https://github.com/ozel/DIY_particle_detector/wiki/Diodes#pin-diodes
or, alternatively, put a thick black piece of cloth on top of all: detector and specimen.

Even with a metal foil glued on top, the diode should be always facing down and away from sources of light.
Too much light saturates the amplifier completely and procues a flat line on the output. 
The tricky condition is with just a little bit of stray light, that keeps the detector running in principle but also raises the noise floor quite a bit. 
One irritating aspect in your oscilloscope picture above is the fact that the left side is flat and the noise waveform starts mid-way only after the trigger. Usually, electromagnetic noise or too much stray light would produce continuous waveforms.
So maybe, in addition, one of the capacitors close to the first amplifier stage is missing or has the wrong value?
Or could the diode be picking up some additional light from the oscilloscope in the moment of the trigger?

Keep me posted.

Matthias32, 2020-10-08:
Hi Oliver,
no the battery is inside the case. For the measurement I put the case with the uranium glass below the aluminum foil. The short cable was used for another setup using a little portable scope. But the big scope in the picture has a better resolution for sure. 

Where I can get these thin metal foils from? Or how should I disassemble the foil capacitors to get the foil?

ozel, 2020-10-09:
> Where I can get these thin metal foils from? Or how should I disassemble the foil capacitors to get the foil?

Many types of foil or [film capacitors](https://en.wikipedia.org/wiki/Film_capacitor) should work as long as there is no electrolytic liquid chemistry inside - non-polar film capacitors should be mostly dry inside. I used those old types from an old electronic device (pre 1980 I would say) which seems to have a kind of "ceramic" layer or paint around them. I simply cut it open with strong pliers. If I find the pictures again, I'll post one. 

Did you have some luck with thick black cloth on top of everything? That should really work as long as there is not even the smallest gap between the table surface and cloth allowing light leaking inside.

Matthias32, 2020-10-10:
Dear Oliver,
I have tried to disassemble some capacitors and it went very well. Thank you a lot! But I still have some strange issues. Let me try to explain (which is not easy for me):

1. If I use the thin foil I get a nice low noise signal which looks good. I’m able to get some beta particles but no alpha particles. I guess it is because the foil is still too thick for alpha particles?
[image]
[image]

2. If I put some cloth on the assembly with no additional radiation window (no foil) I see a lot 50 Hz noise. Now if I touch the housing with my hand the noise gets suppressed.
[image]

3. If use just a thick black tape on the radiation window I have a clear low noise signal. If I put anything on it (my hand, probe, foam, any other material) the 50 Hz noise is again there. While touching the metal housing the noise disappears. 
[image]
[image]

Looks like some ground or capacitive issue? What do you think?

Somehow the previous aluminum foils setup suppressed most of the now occurring 50 Hz noise. By the way I have turned off nearly all lights in my cellar workshop, only using an LED tube.

So is this noise caused by EMV or a 50 Hz driven light source? I have no idea…

ozel, 2020-10-14:
Hi Matthias,

thanks for your many pictures and explanations, it makes it much easier to follow!
This is a very interesting situation I had not observed so far. I am guessing that you have/had a combination of light and "tethered" EM noise issue. The waveform shape from an incandescent light bulb would look exactly the same like EM coupling from the mains supply as both use the same 50Hz frequency. But the observed amplitudes of both effects should be different of course.

I guess what you observe in 2) and probably as well 3) is leakage current coming from the main power (you'd short circuit the tiny leakage current on your mains' earth connection by touching the case). The ground/shield connection of all BNC sockets on mains-powered oscilloscopes is usually directly connected to the earth of the wall outlet (via the so called "Schuko" in Germany). If your type of wall socket allows that, you could try turning the oscilloscope power connector by 180 degrees and with some luck the leakage current could drop depending on the internals of the power supply inside the oscilloscope. But the source of leakage current on the earth connection could be as well some other high-power device on the same mains network like a fridge or kitchen stove in other rooms (up to 2 mA of AC voltage are allowed to be disposed by a power-hungry device on earth connections and per device if I remember correctly).
So the perfect way to test this theory is using either some USB powered oscilloscope & computer (without mains power) or just the headset input/soundcard of a laptop or smartphone. Or even better, in one of your gallery pictures you had a battery-powered portable oscilloscope - that would be best!

Now the question is: how does the leakage current couple into the detector circuit? 
Is it possible that the metal case of the diode is touching your aluminium case? Maybe just barely - the thin oxide layer on aluminium could play a role here. If both are in good low-resistance contact the circuit should not work because the diode case is the sensitive cathode and must not be grounded accidentally via the grounded metal case. But if diode and aluminium case are close together - almost touching or separated only by a thin layer of aluminium oxide - the leakage current could couple capacitively into the amplifier via the diode case, as you suggested.

I had assembled an alpha detector similar to your approach about a year ago. Unfortunately I gave it away, so I can't confirm, but I think that I had isolated the diode with some tape and then pressed it into the hole of the aluminium case. The purpose of the tape would be that aluminium case and diode are electrically well separated. The type of tape could even play a role with different permeability and electrostatic properties... could be worth trying out few different ones if the first one doesn't work right away.

What is your alpha source by the way? It is shielded/encapsulated well, the emitted alphas could loose already most of their energy before they reach the detector, irrespectively of the additional capacitor foil.
What is the amplitude of your largest pulses in 1:1 input attenuation config and at 1 Mega ohm impedance of the scope settings?
Oh and when you touch the case in situations 2) and 3) and the noise gets canceled, can you still observe pulses? 

Let me know how it goes.

Matthias32, 2020-10-25:
Hi Oliver,

I'm still trying to figure it out: I have tried to rotate the power connector 180 degree and also used my battery powered mini scope. The issue is still there. The housing of the diode is not touching the case but the distance is very small.

I took the PCB out of the case to make sure there is no capacitance between the diode housing and the case. The only difference is that the noise signal is now a bit stronger. But again if I touch some GND via the signal disappears (while having the diode covered against light). 

I'm using some uranium glass. I just put some foam around to prevent it from rolling away all the time. The bottom side which heads to the detector is not covered.
[image]

I think I can still observe pulses when I touch the case but let me test it again to make sure.

ozel, 2020-10-28:
ok, at least we can address why you don't see alpha pulses: Uranium glass is usually not a noticable source of alpha particles. It should be in theory but in practice, the uranium concentration is so low, that you'd need to be very lucky to find a spot right on top of the surface of the glass where atoms decay and related alphas are not already absorbed by the glass. So far I have only found thoriated glass from flea markes, like these old camera lenses, to actually emit alpha partciles (besides a lot more of beta and gamma, because those lenses can be very active): https://camerapedia.fandom.com/wiki/Radioactive_lenses 
Thoriated lenses can be identified by their tainted, yellowish/brownish glass. An effect of the strong radioactivity...

Besides old smoke detectors with Am-241 sources inside (sometimes sold as "ionization chamber", *cough* Aliexpress *cough*), an easier to find and harmless kind of everyday alpha source is old ceramics featuring uranium glaze: https://www.google.com/search?q=uranium+glaze. 
Most common is this fire-red/orange kind of color. 

A less intensive alpha source, but freely available everywhere, can be collected from radon in the air of any regular room: https://iopscience.iop.org/article/10.1088/0031-9120/32/2/016/meta
In short: attach an eletrostaically charged party ballon for 15 minutes with sticky tape on a wall in a closed room (cellars work best). Release the air slowly out of the balloon without touching the surface by cutting into the balllon at the knot (if you hold the plastic firmly, it will not explode).
Then put the part of the balloon, which is opposite to the knot directly on to of the diode (but keep the thin radiation window capacitor foil in between!). 

Regarding the noise problem:
I am suprised that you see the same noise effect with the battery powered oscilloscope. That either means there is too much light reaching the diode or that you have significant EM transmission in the air around your house (or the oscilloscope itself is too noisy for these low signals or are picking up EM themsleves - but unlikely if both oscilloscpe show exactly the same noise patern. Maybe try it in a completley different building with the battery powered setup? A garden shed or outside in a *dark corner* would be ideal.

If I undestand correctly, having the case around the circuit does not make much of a difference. From this I would conclude that the ground connection between circuit and the case is not sufficient. You should have at least a short GND wire coming from PCB and soldered to the BNC connector. If that one is firmly mounted in the case it should be fine. In addition, I guess you are using at least one or two metal screws that hold the PCB in place through either the three large PCB holes or the two exposed PCB edges which are all grounded.
Could you maybe record a short video using the the batery-powerd oscillsocpe?

Looking at the uranium marble and  foam, I hope you don't expect that it is blocking the light well enough. The alpha spectrometer circuit is not yet as sensitive as a single photon-detector but really rather close to that. The light coming through the foam and marble would be by far too much if either the diode or the whole detetor setup is not kept in complete darkness.

Matthias32, 2020-10-31:
Dear Oliver,

thank you a lot for the hint. I found some uranium glaze cups on eBay. I will order them and try again. 

I changed the location but the issue was similar (see below). 

No my GND is not floating: I have soldered a wire between GND and the BNC connector and also added metal screws which connect to the two GND planes at the PCB corners. Outside the housing I have at least a wire between GND and BNC connected.

It is very strange: I have covered the diode completely with black tape and did some measurements even in nearly complete darkness. For this test I’m using a fresh diode which has still the glass cover on it so the tape can cover better (no little holes). If get closer to my working bench the noise gets worse. (Below the wooden desk there is a lot of metal) If I use the mobile scope while walking in the house the noise signal changes a lot. If I come close to some metal it gets worse. At some places the noise is nearly 0 regardless if the light is switched on or off.

[image]
(I did the test in the video also in nearly complete darkness. The result was the same)

Anyway If the detector is inside the case with the metal foil covering the sensor hole it works fine (see my post from 10.10) The noise is very low and I’m able to detect beta particles. When the uranium glaze cups arrive I will test again for alpha particles.

So is it just EM noise? But why makes it such a difference if the sensor hole is covered with a metal foil or just black tape? I mean the hole is way too small for 50 Hz noise passing. On the other side the effect seems to be very sensitive to any capacitive change.

ozel, 2020-11-03:
Thanks for the video! :-) 
But honestly, it disturbs me a little bit. The detector must be always in a metal enclosure surrounding the whole circuit board. I thought this was clear and valid in all your previous tests, even when there was the dark blanket on top! 
The metal is shielding against light and most forms of electromagnetic noise like a Faraday cage. That is its purpose, otherwise, the circuit is just a somewhat expensive capacitive touch/proximity sensor.

The T0 metal case of the diode itself does not really contribute to EM shielding. The whole first amplifier stage is extremely sensitive to charges and stray capacitances.

Regrading the frequency properties, there are different factors at play. EM noise can couple in different ways. I'd recommend to check out the literature on EMI for product development (German 'EMV'), it's a fascinating but really complex topic...

Matthias32, 2020-11-06:
No you don’t understand: I had the issue with and without housing (see one of my previous posts). I took it out of the housing to check if the capacitance between the diode housing and the case is an issue as you mentioned before. Also I was able to do better debugging without housing.

Don’t worry I know well about EMI since it is my daily business 😉 

Anyway: I resoldered all of the components, added a better connection to the BNC plug (it was not ideal before) and cut down the diode housing a bit to make sure there no short circuit after assembly. Together with a fresh sensor opening foil and the new uranium glace probe I finally receive pulses like this:
[image]

What do you think?

ozel, 2020-11-06:
Congratulations! 
The pulse amplitude looks well bigger than the -8 mV threshold which, if measured at regular 1 Megaohm input impedance and 1:1 input attenuation, renders it a full-fledged alpha particle pulse.
c.f. https://github.com/ozel/DIY_particle_detector/wiki/Oscilloscope-Measurements

Looking at your upper video again, it could be dominated by fluorescent/LED room light - that is bouncing back from the table surface as you approach it - and causing the strong 50 Hz waveforms. Generally, one layer of black insulation tape is often not enough for blocking all the light in a well-lit room and depending on the tape.

The main difficulty is that "mains clocked" light and other mains powered EM coupling may induce similar sine waves such that both may be contributing at any time if not shielded enough individually.

closed 2020-12-09 by ozel