ozel/DIY_particle_detector #15: MALAS DETECCIONES DETECTORES ALFA Y BETA

josepanglada6, 2020-12-03:
Hola buenas,
Compré los dos dispositivos (alfa y beta) y monté las placas electrónicas y hice una caja de aluminio con una ventana de radiación que se protege con cinta adhesiva metálica.
Pero cuando abro un programa como Becqmoni o alguna página web de osciloscopio, solo me marca detecciones cuando doy golpes en la placa o en el cable de sonido. Si acerco elementos radioactivos no hace nada. A qué puede ser debido esto? Que programas se recomiendan? Es necesario colocar otro tipo de cinta en la ventana de radiación?
Gracias.

ozel, 2020-12-08:
Hello, 
I had to translate the question since I do not speak Spanish.
Please consider posting in English.

To get your detectors running, please have a look at the Troubleshooting page in the wiki:
https://github.com/ozel/DIY_particle_detector/wiki/Troubleshooting
If you use the audio/microphone input of a Laptop, you probably have to change the microphone volume setting in Windows. Sometimes with headset connectors, it should be between 50% and 75% to work properly, this depends on your computer.
Also check that you have the correct cable for your computer:
https://github.com/ozel/DIY_particle_detector/wiki/Cables#audio-jack-connector-overview-35-mm-trstrrs
If you use the same low-cost soundcard as me, the CM 108, the input volume should be set at 100% with extra microphone boost/gain enabled:
https://github.com/ozel/DIY_particle_detector/wiki/Soundcards

josepanglada6, 2020-12-09:
Thanks for the interest Ozel. 

I have two questions for the alpha detector:
When I use the application Becqmoni in order to study the detections, if I apply the formula that is said on the web for the energy calibration, I get some detections with negative energy. Is this correct?https://github.com/ozel/DIY_particle_detector/wiki/Energy-Spectra#energy-calibration 

On the other hand, I have no idea how to use this python script: https://github.com/ozel/DIY_particle_detector/blob/master/data_analysis_and_reference_measurements/diode_detector/analyse_and_plot_pulses.py 
I have downloaded the program annaconda. Then, it represents that I have to copy all the python script adding on it the file name and something else? I have to copy all the script or I have to delet the clarifications, the instructions of the beggining...? And where do I have to copy the script? 
If you could explain me how this works I would appreciate it.

Thanks, Josep.

ozel, 2020-12-09:
Hi Josep,

I have split your questions up into two posts in the brand new Discussions feature of Github!
Please follow these links: #18 #20 
I will close this issue later.

Cheers,
Oliver

closed 2020-12-14 by ozel